odya/esphome-powmr-hybrid-inverter #1: Voltage control

tomkolp, 2023-06-10:
I have POW-HVM1.5H-12V, will it be possible to control the voltages ever?

odya, 2023-07-06:
Sorry, I have only 24v model. What voltages are you talking about?

odya, 2023-07-07:
For now it were tested only on 2.4kW/24v model. I added a manuals section because during my research, I accumulated various manuals. I assumed that the 1.5, 2.0, 2.4, and 3.2 models are similar because they have mostly similar manuals. I apologize for any confusion.

Paxy, 2023-09-05:
Have you tried to make a switch instead of the sensor for writing Modbus value instead of just reading it? This could be useful for remote voltage control, power source selection, ...
Do you have yaml sample with switches that works?

odya, 2023-09-06:
There are separate registers for reading and writing. I am using the [modbus_select](https://esphome.io/components/select/modbus_controller) component for changing write registers. For binary values, you can use the Modbus Switch component instead.