odya/esphome-powmr-hybrid-inverter #11: POW-HVM3.2H-24V modbus write registers

AugustoMoura, 2023-10-03:
Hi
I just want to share some extra write registers that I found in my POW-HVM3.2H-24V

I used an ESP32 and this https://github.com/zivillian/esp32-modbus-gateway/
By using the debug menu: "Debug" -> Slave ID 5 + "Read Holding Register" + Register # -> "Send" and comparing with the settings in the inverter screen
 
Here's the list: (just be careful, I haven't fully tested it)

- 5002 : Buzzer Alarm (range 0-1, settings menu 18)
- 5004 : Backlight control (range 0-1, settings menu 20)
- 5005 : Auto restart when overload occurs (range 0-1, settings menu 6)
- 5006 : Auto restart when over temperature occurs (range 0-1, settings menu 7)
- 5007 : Beep On Primary Source Fail (range 0-1, settings menu 22)
- 5008 : Auto return to default display screen (range 0-1, settings menu 19)
- 5009 : Overload Bypass (0-1, settings menu 23)
- 5010 : Record fault code (range 0-1, settings menu 25)
- 5017 : Charger Source Priority (range 0-3, settings menu 16)
- 5018 : Output Source Priority (range 0-2, settings menu 1)
- 5019 : AC input voltage range (range 0-1, settings menu 3) (0 - 90-280VAC, 1 - 170-280VAC)
- 5020 : Battery type (range 0-2, settings menu 5)
- 5021 : Output frequency (range 0-1, settings menu 9) (0 - 50hz, 1 - 60hz)
- 5022 : Max Total Charge Current (range 10-80, settings menu 2)
- 5023 : Output voltage (one of 220, 230, 240, settings menu 10)
- 5024 : Utility Charge Current (one of 2, 10, 20, 30, 40, 50, 60, settings menu 11)
- 5025 : Comeback Utility Mode Voltage (SBU) (0.5 volts step, settings menu 12) 
- 5026 : Comeback Battery Mode Voltage (SBU) (0.5 volts step, settings menu 13)
- 5027 : Bulk charging voltage (settings menu 26)
- 5028 : Floating charging voltage (settings menu 27)
- 5029 : Low DC cut-off voltage (settings menu 29)
- 5030 : Battery equalization voltage (settings menu 31)
- 5031 : Battery equalized time (settings menu 33)
- 5032 : Battery equalized timeout (settings menu 34)
- 5033 : Equalization interval (settings menu 35)

romanbychkoosf, 2023-10-04:
thanks for pointing to modbus-gateway!
With it I identified temperature sensor of MPPT controller of POWMR 3.6M

- platform: modbus_controller
    modbus_controller_id: smg_inverter
    name: "PV Temperature"
    address: 4557
    register_type: holding
    value_type: U_WORD
    unit_of_measurement: "c"
    device_class: temperature
    state_class: measurement
    accuracy_decimals: 1
    lambda: |-
      return swapBytes(x);

odya, 2023-10-10:
@AugustoMoura thanks. Updated registers map https://github.com/odya/esphome-powmr-hybrid-inverter/pull/17

closed 2023-10-10 by odya