odya/esphome-powmr-hybrid-inverter #23: Duplicate modbus command found

davidebeatrici, 2023-10-28:
```
[W][modbus_controller:136]: Duplicate modbus command found: type=0x3 address=4502 count=38
[W][modbus_controller:136]: Duplicate modbus command found: type=0x3 address=4538 count=1
[W][modbus_controller:136]: Duplicate modbus command found: type=0x3 address=4540 count=16
```

According to [ESPHome's code](https://github.com/esphome/esphome/blob/d4cb29a38050b1dd5c838a60b689f406753c058b/esphome/components/modbus_controller/modbus_controller.cpp#L130-L144) the warning is logged when the exact command is already in the queue, which I guess may happen in our case because we are reading quite a lot.

However, I'm wondering why only those three addresses are affected and in a consistent manner.

Consider that I had to add `command_throttle` to the `modbus_controller` section (present in the old example), otherwise all retrieved values happened to be garbage.

davidebeatrici, 2023-10-28:
This is with an Anern AN-SCI-MAX-10200, almost identical to the PowMr POW-HVM10.2M.

odya, 2023-11-01:
1) Do you have updated latest version (1.3.2) of firmware? I did some fixes with modbus addresses gaps & optimisations for selects update frequency in latest patches.
2) Have you checked your modbus addresses for gaps?
3) What update interval you have on modbus sensors? I have 30s interval now but on 20s I didn't receive such errors too.

davidebeatrici, 2023-11-01:
1. Yep. I actually got rid of the selects because I don't need them.
2. What do you mean?
3. 2s as I would like at least the power consumption reading to update as quickly as possible. 

Increasing the update interval to 5s fixed the issue, thanks! I didn't expect this to be the cause as the warning is only emitted for 3 addresses...

Please note that setting `command_throttle` to at least 1s is still required, otherwise:

```
[22:51:52][W][modbus:108]: Modbus CRC Check failed! B7CB!=1965
[22:51:52][W][modbus:108]: Modbus CRC Check failed! 4289!=FB9B
[22:51:53][W][modbus_controller:030]: Modbus device=5 set offline
[22:51:53][W][modbus_controller:064]: Modbus device=5 back online
[22:51:57][W][modbus:108]: Modbus CRC Check failed! 861C!=8AC9
[22:51:57][W][modbus:108]: Modbus CRC Check failed! 95A0!=4488
[22:51:58][W][modbus_controller:030]: Modbus device=5 set offline
[22:51:58][W][modbus_controller:064]: Modbus device=5 back online
[22:52:02][W][modbus:108]: Modbus CRC Check failed! 8082!=44F8
[22:52:02][W][modbus:108]: Modbus CRC Check failed! 30B4!=BBD7
[22:52:03][W][modbus_controller:030]: Modbus device=5 set offline
[22:52:03][W][modbus_controller:064]: Modbus device=5 back online
[22:52:07][W][modbus:108]: Modbus CRC Check failed! E0E5!=68D3
[22:52:07][W][modbus:108]: Modbus CRC Check failed! 564F!=F471
[22:52:08][W][modbus_controller:030]: Modbus device=5 set offline
[22:52:08][W][modbus_controller:064]: Modbus device=5 back online
[22:52:12][W][modbus:108]: Modbus CRC Check failed! CA52!=9F2E
[22:52:12][W][modbus:108]: Modbus CRC Check failed! 6EA6!=5C44
[22:52:13][W][modbus_controller:030]: Modbus device=5 set offline
[22:52:13][W][modbus_controller:064]: Modbus device=5 back online
```

Should I open a pull request?

odya, 2023-11-02:
2) ESPHome, by default, requests sequential registers in one batch. That greatly optimizes throughput.

3) I tested it on my configuration (with selects). I was only able to eliminate these messages with a 10s interval. `command_throttle` did not have any significant effect for me. Updating the interval resolved the issue. However, without selects, I believe you could have smaller intervals because each select uses a separate Modbus range (this was a bug fix for `Modbus device=5 set offline` warnings).

> Should I open a pull request?
In any case I will set `command_throttle` to 1s today in new version branch. Found some minor bugs while testing😀

davidebeatrici, 2023-11-02:
Alright, thanks!

closed 2023-11-09 by odya