odya/esphome-powmr-hybrid-inverter #27: Compiling for esp32c3

bronzeblood, 2023-12-06:
What parameters need to be changed to use it with the advanced pcb that @davidebeatrici designed. I can get it to compile but the esp just boot loops.

bronzeblood, 2023-12-06:
ok i got it to compile by replacing this..

```
esp32:
  variant: ESP32C3
  board: esp32-c3-devkitm-1
  framework:
    type: esp-idf
```
Seems like my boot loop was caused by low voltage from the usb uart i was using, when powered from the inverter its starting fine. 

However i'm not getting any communication, can someone clarify the gpio pins used for tx/rx on the advanced design.

bronzeblood, 2023-12-06:
Figured it out 

```
  inverter_tx_pin: GPIO7
  inverter_rx_pin: GPIO6
```

When using the advanced pcb

All seems to be working fine, maybe this will help anyone that makes the advanced pcb in the future.

davidebeatrici, 2023-12-06:
Thank you for using my advanced PCB!

For reference, it was added in #10.
Please note that I've been using `arduino` as framework type without setting the flash mode, but it's probably more correct to use `esp-idf` since it's the official Expressif framework.

Would you like to create a pull request to add the instructions to https://github.com/odya/esphome-powmr-hybrid-inverter/blob/063f91e16edf12ee3f00fc5a1b3e03f57f2eb2c1/pcb/advanced/README.md?