odya/esphome-powmr-hybrid-inverter #31: PV Power from POW-HVM10.2M seems to only show one of the two PV string

pkirk, 2024-03-25:
Gentleman,
I have two PV strings to my inverter(4 + 4kW) , but the register 4505 seems to show only one. The same thing is on the physical display of the Inverter, where only one string at a time is showed, but clicking make it cycle and show first one and then, after some other screen, the other one.

Any idea?

odya, 2024-04-13:
Unfortunately, I don't have PV panels, so I can't test whether the values from them are accurate. However, I think there might be some additional register on more powerful inverter models.

bradmatt275, 2024-05-20:
Through trial and error I'm pretty sure I have found the registers on my POW-HVM10.2M:

```
  - platform: modbus_controller
    modbus_controller_id: smg_inverter
    name: "PV 2 Voltage"
    id: pv2_voltage
    address: 4563
    register_type: holding
    value_type: U_WORD
    unit_of_measurement: "V"
    device_class: voltage
    state_class: measurement
    accuracy_decimals: 1
    lambda: |-
      return swapBytes(x);
    filters:
      - multiply: 0.1

  - platform: modbus_controller
    modbus_controller_id: smg_inverter
    name: "PV 2 Power"
    id: pv2_power
    address: 4564
    register_type: holding
    value_type: U_WORD
    accuracy_decimals: 1
    unit_of_measurement: "W"
    device_class: power
    state_class: measurement
    lambda: |-
      return swapBytes(x);

  - platform: template
    name: "PV 2 Current"
    state_class: "measurement"
    device_class: current
    unit_of_measurement: "A"
    accuracy_decimals: 1
    icon: mdi:solar-power
    lambda: |-
      return id(pv2_power).state / id(pv2_voltage).state;
```

[image]

pkirk, 2024-05-21:
Thanks @bradmatt275 but, too bad, my inverter doesn't have those values :(

bradmatt275, 2024-05-21:
> Thanks @bradmatt275 but, too bad, my inverter doesn't have those values :(

Oh I guess they must have updated the firmware. Mine is quite new and it has the new display.

Well not that it matters now. After weeks of work I had it all working perfectly. Then I ruined it by mis wiring the BMS cable and fried the comms card.

davidebeatrici, 2024-06-19:
Unfortunately the registers don't seem to contain anything on my Anern AN-SCI-MAX-10200 that I bought in September 2023 either.

@bradmatt275 Do you see both strings on your inverter's display?

bradmatt275, 2024-06-19:
> Unfortunately the registers don't seem to contain anything on my Anern AN-SCI-MAX-10200 that I bought in September 2023 either. [...]

Yes I see both of them. I ended up getting a replacement comms card (I burnt my first one) and confirmed the display matches what I'm seeing from ESP Home / HA for both trackers.

The only difference is for some reason at night, when the inverter display shows zero volts for both trackers. HA will read 45v and 0.01 amps.

I'll have to test it one day with my multimeter, but I know it is possible for solar arrays to have small amounts of voltage at night. It might be that the display doesn't register it until it gets close to the start up voltage.

With that said u have a different model to yours so the registers might be different.

davidebeatrici, 2024-06-21:
> Yes I see both of them. I ended up getting a replacement comms card (I burnt my first one) and confirmed the display matches what I'm seeing from ESP Home / HA for both trackers.

Could you leave here a picture of your display, just for reference?

> I'll have to test it one day with my multimeter, but I know it is possible for solar arrays to have small amounts of voltage at night. It might be that the display doesn't register it until it gets close to the start up voltage.

That's probably the case, but I would check just in case.

> With that said u have a different model to yours so the registers might be different.

Before switching to this model (which is very similar) I had a POW-HVM10.2M and its display only showed one string.

patogomaII, 2024-06-21:
> Unfortunately the registers don't seem to contain anything on my Anern AN-SCI-MAX-10200 that I bought in September 2023 either. [...]

I have the same inverter and registers 4563 and 4564 shows me PV2 correct data. I'm using stock wifi dongle reprogrammed in Home Assistant and all works fine except select fields.
In the Anern's screen, you can see PV2 data pushing down arrow. Data in the down-left corner shows PV1 and PV2 data secuently.

davidebeatrici, 2024-06-21:
May I ask when you bought the inverter?

bradmatt275, 2024-06-22:
> Could you leave here a picture of your display, just for reference?
>
>
> That's probably the case, but I would check just in case.
>
>
> Before switching to this model (which is very similar) I had a [...]

Sure. There are some photos on the screen and my readings on HA. They are slightly off because of the delay of taking the photos and the screenshot. But still within 0.01a.
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

davidebeatrici, 2024-06-22:
Thank you.

I find it quite interesting how PowMr switched to a screen that is the exact same as on the Anern but otherwise kept the previous style (capacitive buttons rather than tactile).

For reference, this is how the screen on my POW-HVM10.2M looks like:

[image]

Seems like I'll have to ask for a firmware upgrade to get both PV strings to show up...

bradmatt275, 2024-06-22:
> Thank you.
>
> I find it quite interesting how PowMr switched to a screen that is the exact same as on the Anern but otherwise kept the previous style (capacitive buttons rather than tactile).
>
> For [...]

Yeah it was a surprise when I got it. All their videos and manuals had the old style screen. My previous POW-HVM6.2M also has the same screen as yours.

patogomaII, 2024-06-22:
> May I ask when you bought the inverter?

I bought it at 2023 november. At the end of the secuential menu in your screen, you can see the firmware version. What yours?

bradmatt275, 2024-06-22:
> > May I ask when you bought the inverter?
> 
> I bought it at 2023 november. At the end of the secuential menu in your screen, you can see the firmware version. What yours?

Sorry I have no idea how to check the firmware version. The only reading that didn't look like something power related was U1 at the end. Not sure if that is a version number.

But I can confirm I ordered it on April 2024.

Also I'm not sure if it makes any difference with the modbus readings (I suspect not), but I don't have the original comms card in there. I took it from my old POW-HVM6.2M because I fried my original one that came with the unit. Although the PCB/components looked exactly the same.

patogomaII, 2024-06-22:
Yes, U1 is the firmware version. Send a photo of this.

bradmatt275, 2024-06-22:
> Yes, U1 is the firmware version. Send a photo of this.

[image]

patogomaII, 2024-06-22:
Wow! 79.05
Mine is 54.01

bradmatt275, 2024-06-23:
> Wow! 79.05 Mine is 54.01

I'm curious how you upload new firmware. There is no USB on these models like some of the others.

davidebeatrici, 2024-06-23:
Turns out the display on my Anern does indeed show both strings, I just never noticed because they perform very similarly:

[image]
[image]

My firmware is the oldest so far though:

[image]

Please ignore the fault condition, it's because the battery is close to the low voltage limit I set.

patogomaII, 2024-06-23:
There's no firmware update neither support via Anern official site. I ask for it and only received the silence of the inmense ciberspace...

Tibi851026, 2024-09-15:
Hello. Anyone can help me I would like to upgrade my firmware because I have same problem. But I can not find any firmware and the program what have to use for the firmware upgrade. Thanks!

bradmatt275, 2024-09-16:
> Hello. Anyone can help me I would like to upgrade my firmware because I have same problem. But I can not find any firmware and the program what have to use for the firmware upgrade. Thanks!

You might have to contact their support on the website. I don't think they make it publicly available. 

They are actually pretty quick to reply most of the time. Although the language barrier can be a little difficult.