odya/esphome-powmr-hybrid-inverter #43: Unavailable daily energy sensors

dmode9000, 2024-05-21:
Hi Andrii,
Thank you for your invertor integration.
Tried to use sunsynk-power-flow-card with PowMr invertor [according to the example](https://slipx06.github.io/sunsynk-power-flow-card/examples/powmr.html), but daily/total energy sensors are unavailable for PV and battery. 

- sensor.powmr_inverter_pv_yield_daily
- sensor.battery_charge_daily

[image]

Sunsynk-power-flow-card developer [@slipx06 says](https://community.home-assistant.io/t/inverter-power-flow-card/562933/390) the problem is with the integration. I would appreciate any help in fixing this issue.

Nmebowen, 2024-05-31:
best post your yaml but my best guess is the example yaml doesnt actually correspond to the main/inverter yaml names. if you alter the card yaml manually and under those sections all just the last part ie pv_yield the corresponding sensor should show up

gloryib, 2024-08-04:
You need create a sensors in helpers to count them.