odya/esphome-powmr-hybrid-inverter #47: POW-HVM4.5K-24V/POW-HVM6.5K-48V support

lordlightman, 2024-06-10:
Hello there.

Does anyone know if newer PowMr inverters like POW-HVM4.5K-24V/POW-HVM6.5K-48V are supported?

lufton, 2024-06-13:
@lordlightman no, but you can check my repository for your inverter, it should work: https://github.com/lufton/esphome-inv-8851

lordlightman, 2024-06-14:
Hello @lufton, thanks, I will check your repo.

PZotya, 2024-10-15:
Leodesigner made a project for the communication basics, which uses the inv8851 also. In this documentation he emphasizes, that the setting modification is available only after requesting a config, and sending it back the modified one. Does it can work with this project?
Leodesigner's project for the POW-HVM4.5K-24V/POW-HVM6.5K-48V inverter communication: https://github.com/leodesigner/powmr4500_comm/tree/main

lufton, 2024-10-15:
@PZotya, my project works perfectly with setting  parameters. The way it works it periodically requests state, then replaces parameter bit and pushing this config back.

PZotya, 2024-10-19:
> @PZotya, my project works perfectly with setting parameters. The way it works it periodically requests state, then replaces parameter bit and pushing this config back.

Hi Lufton!

Thanks for this good work!
I was able to use Your ESPHome setup for my PowMr inverter and HomeAssistant. But I have a problem, that all my entities of this device is Unknown. What is that I made wrong?

lufton, 2024-10-19:
@PZotya please open corresponding issue in my project. Specify witch hardware you use and your setup details including software version of inverter.

closed 2024-10-19 by lordlightman