odya/esphome-powmr-hybrid-inverter #48: Charger source priority and Output souce priority always returns to value "utility first" and "USB"

patogomaII, 2024-06-21:
I don't understand why this happens, but even if I change the values ​​to "only solar" and "SBU priority", they immediately change the values ​​back to "Utility first" and "USB priority". Even if I change them on the inverter screen directly.
I'm using an Anern AN-SCI-MAX 10.2kw and the wifi dongle (esp8266).

Nando1497, 2024-06-29:
I have the same problem as you but it doesn't do it immediately after several hours

Nando1497, 2024-06-30:
Hi, I solved it by commenting the code below. once deactivated, take the configurations given to the inverter.

#  - platform: modbus_controller
#    modbus_controller_id: smg_inverter
#    id: charger_source_priority_text
#    name: "Charger Source Priority"
#    entity_category: diagnostic
#    address: 4536
#    register_type: holding
#    response_size: 2
#    raw_encode: HEXBYTES
#    lambda: |-
#      uint16_t sensorIndex = swapBytes(modbus_controller::word_from_hex_str(x, 0));
#      updateUnknownSelect(sensorIndex, id(charger_source_priority_select));
#      switch (sensorIndex) {
#        case 0: return std::string("Utility first");
#        case 1: return std::string("Solar first");
#        case 2: return std::string("Solar and Utility");
#        case 3: return std::string("Only solar");
#        default: return std::string(x);
#      }

#  - platform: modbus_controller
#    modbus_controller_id: smg_inverter
#    name: "Output Source Priority"
#    entity_category: diagnostic
#    address: 4537
#    register_type: holding
#    response_size: 2
#    raw_encode: HEXBYTES
#    lambda: |-
#      uint16_t sensorIndex = swapBytes(modbus_controller::word_from_hex_str(x, 0));
#      updateUnknownSelect(sensorIndex, id(output_source_priority_select));
#      switch (sensorIndex) {
#        case 0: return std::string("Utility first (USB)");
#        case 1: return std::string("Solar first (SUB)");
#        case 2: return std::string("SBU priority");
#        default: return std::string(x);
#      }

#  - platform: modbus_controller
#    id: charger_source_priority_select
#    name: "Charger Source Priority"
#    optimistic: true
#    skip_updates: ${select_skip_updates}
#    force_new_range: true
#    entity_category: config
#    address: 5017
#    value_type: U_WORD
#    optionsmap:
    
#      "Solar and Utility": 2
#      "Solar first": 1
#      "Utility first": 0
#      "Only Solar": 3

#  - platform: modbus_controller
#    id: output_source_priority_select
#    name: "Output Source Priority"
#    optimistic: true
#    skip_updates: ${select_skip_updates}
#    entity_category: config
#    address: 5018
#    value_type: U_WORD
#    optionsmap:
    
#      "SBU priority": 2
#      "Solar first (SUB)": 1
#      "Utility first (USB)": 0

patogomaII, 2024-06-30:
Yes. I do that too, but i'd like to have the capacity of change that options, remotely and via Home Assistant, in order to integrate that options in various automations.

lufton, 2024-09-02:
Inexperience the same issue too. @odya could you please help here?

volodymyr-fedorchuk, 2024-09-02:
+1, the same issue. The charger source priority can be changed itself w/o any trigger, usually it happens several times per day. From logs:

[image]

davidebeatrici, 2024-09-08:
In my case "Output Source Priority" shows up as "SBU priority", which matches the inverter's actual setting.

However, "Charger Source Priority" shows up as "Utility first", which is wrong. The inverter's actual setting is "Solar first".