odya/esphome-powmr-hybrid-inverter #49: does not read the values ​​"Grid Active, Grid Active, Grid Power, Grid Voltage"

Nando1497, 2024-06-24:
Hi, I can't read the "Grid Active, Grid Active, Grid Power, Grid Voltage" values. I have an esp32: M5Stack Atom Lite ESP32.
[image]

Andrik45719, 2024-06-29:
Same problem, solved by change yaml to read states from 4553 register instead 4554

  - platform: modbus_controller
    modbus_controller_id: smg_inverter
    name: "Grid Active"
    id: grid_active
    address: 4553
    register_type: holding
    bitmask: 0x200

  - platform: modbus_controller
    modbus_controller_id: smg_inverter
    name: "On Battery"
    address: 4553
    register_type: holding
    bitmask: 0x100

Nando1497, 2024-06-29:
hi, changing from 4554 to 4553 I read Grid Active and On Battery, but Grid Power not yet

Nando1497, 2024-06-29:
It often happens that the "output Source Priority" and the "charger Source Priority" change

davidebeatrici, 2024-09-08:
> It often happens that the "output Source Priority" and the "charger Source Priority" change

Related: #48

rublin, 2024-12-06:
I have the same issue with my POW-HVM10.2M. 
Fixed **Grid Active** and **Grid Voltage** with this configuration in `inverter.yaml`:

```yaml
  - platform: modbus_controller
    modbus_controller_id: smg_inverter
    name: "Grid Active"
    id: grid_active
    address: 4553
    register_type: holding
    bitmask: 0x2000

  - platform: modbus_controller
    modbus_controller_id: smg_inverter
    name: "On Battery"
    address: 4553
    register_type: holding
    bitmask: 0x100
```
But the **Grid Current** and **Grid Power** still don't work

kevinari26, 2024-12-20:
me too, all value are NA
[image]

balbulator, 2025-01-06:
I noticed that Grid Voltage works and provides data if the inverter output is turned off, but when turned on, it shows 0.  Inverter Codi 6.2 kW

Nando1497, 2025-02-12:
the value "Grid Active " is active with this configuration

  - platform: modbus_controller
    modbus_controller_id: smg_inverter
    name: "Grid Active"
    id: grid_active
    address: 4554
    register_type: holding
    bitmask: 0x200
    
  - platform: modbus_controller
    modbus_controller_id: smg_inverter
    name: "On Battery"
    address: 4554
    register_type: holding
    bitmask: 0x100

[image]