odya/esphome-powmr-hybrid-inverter #50: "Received nan" almost for all commands

sevrugin, 2024-06-29:
Can you help me to figure out the issue, please?

Invertor: POW-HVM6.2M

Microcontroller: M5 Atomo lite + RS232->ttl
Config as in examples

this is the part of log

```
[15:03:24][C][modbus_controller:349]: ranges
[15:03:24][C][modbus_controller:352]:   Range type=3 start=0x1399 count=1 skip_updates=2
[15:03:24][C][modbus_controller:352]:   Range type=3 start=0x1196 count=37 skip_updates=0
[15:03:24][C][modbus_controller:352]:   Range type=3 start=0x11BC count=16 skip_updates=0
[15:03:24][C][modbus_controller:352]:   Range type=3 start=0x138A count=1 skip_updates=2
[15:03:24][C][modbus_controller:352]:   Range type=3 start=0x138F count=1 skip_updates=2
[15:03:24][C][modbus_controller:352]:   Range type=3 start=0x1391 count=1 skip_updates=2
[15:03:24][C][modbus_controller:352]:   Range type=3 start=0x139A count=1 skip_updates=2
[15:03:24][C][modbus_controller:352]:   Range type=3 start=0x139E count=1 skip_updates=2
[15:03:24][C][modbus_controller:352]:   Range type=3 start=0x13A0 count=1 skip_updates=2
[15:03:24][C][modbus_controller:354]: server registers
[15:03:24][V][modbus_controller:044]: Sending next modbus command to device 5 register 0x1399 count 1
[15:03:24][V][modbus:210]: Modbus write: 05.03.13.99.00.01.51.25 (8)
[15:03:24][V][modbus_controller:556]: Command sent 3 0x1399 1
[15:03:24][D][uart_debug:114]: >>> 05:03:13:99:00:01:51:25
[15:03:24][V][sensor:043]: 'Load Power': Received new state nan
[15:03:24][D][sensor:094]: 'Load Power': Sending state nan W with 0 decimals of accuracy
[15:03:25][D][sensor:094]: 'Battery Current': Sending state nan A with 1 decimals of accuracy
[15:03:25][V][modbus_controller:044]: Sending next modbus command to device 5 register 0x1399 count 1
[15:03:25][V][modbus:210]: Modbus write: 05.03.13.99.00.01.51.25 (8)
[15:03:25][V][modbus_controller:556]: Command sent 3 0x1399 1
[15:03:25][D][uart_debug:114]: >>> 05:03:13:99:00:01:51:25
[15:03:26][V][modbus_controller:044]: Sending next modbus command to device 5 register 0x1399 count 1
[15:03:26][V][modbus:210]: Modbus write: 05.03.13.99.00.01.51.25 (8)
[15:03:26][V][modbus_controller:556]: Command sent 3 0x1399 1
[15:03:26][D][uart_debug:114]: >>> 05:03:13:99:00:01:51:25
[15:03:26][V][internal_temperature:034]: Raw temperature value: 118
[15:03:26][V][sensor:043]: 'Controller Temperature': Received new state 47.777779
[15:03:26][D][sensor:094]: 'Controller Temperature': Sending state 47.77778 °C with 1 decimals of accuracy
[15:03:27][W][modbus_controller:027]: Modbus device=5 set offline
[15:03:27][D][modbus_controller:040]: Modbus command to device=5 register=0x1399 countdown=0 no response received - removed from send queue
[15:03:27][V][modbus_controller:044]: Sending next modbus command to device 5 register 0x1196 count 37
[15:03:27][V][modbus:210]: Modbus write: 05.03.11.96.00.25.60.85 (8)
[15:03:27][V][modbus_controller:556]: Command sent 3 0x1196 37
[15:03:27][D][uart_debug:114]: >>> 05:03:11:96:00:25:60:85
[15:03:28][V][modbus_controller:044]: Sending next modbus command to device 5 register 0x1196 count 37
[15:03:28][V][modbus:210]: Modbus write: 05.03.11.96.00.25.60.85 (8)
[15:03:28][V][modbus_controller:556]: Command sent 3 0x1196 37
[15:03:28][D][uart_debug:114]: >>> 05:03:11:96:00:25:60:85
[15:03:29][V][modbus_controller:044]: Sending next modbus command to device 5 register 0x1196 count 37
[15:03:29][V][modbus:210]: Modbus write: 05.03.11.96.00.25.60.85 (8)
[15:03:29][V][modbus_controller:556]: Command sent 3 0x1196 37
[15:03:29][D][uart_debug:114]: >>> 05:03:11:96:00:25:60:85
[15:03:30][V][modbus_controller:044]: Sending next modbus command to device 5 register 0x1196 count 37
[15:03:30][V][modbus:210]: Modbus write: 05.03.11.96.00.25.60.85 (8)
[15:03:30][V][modbus_controller:556]: Command sent 3 0x1196 37
[15:03:30][D][uart_debug:114]: >>> 05:03:11:96:00:25:60:85
[15:03:31][V][modbus_controller:044]: Sending next modbus command to device 5 register 0x1196 count 37
[15:03:31][V][modbus:210]: Modbus write: 05.03.11.96.00.25.60.85 (8)
[15:03:31][V][modbus_controller:556]: Command sent 3 0x1196 37
[15:03:31][D][uart_debug:114]: >>> 05:03:11:96:00:25:60:85
[15:03:32][D][modbus_controller:040]: Modbus command to device=5 register=0x1196 countdown=0 no response received - removed from send queue
[15:03:32][V][modbus_controller:044]: Sending next modbus command to device 5 register 0x11BC count 16
[15:03:32][V][modbus:210]: Modbus write: 05.03.11.BC.00.10.81.5A (8)
[15:03:32][V][modbus_controller:556]: Command sent 3 0x11BC 16
[15:03:32][D][uart_debug:114]: >>> 05:03:11:BC:00:10:81:5A
[15:03:33][V][modbus_controller:044]: Sending next modbus command to device 5 register 0x11BC count 16
[15:03:33][V][modbus:210]: Modbus write: 05.03.11.BC.00.10.81.5A (8)
[15:03:33][V][modbus_controller:556]: Command sent 3 0x11BC 16
[15:03:33][D][uart_debug:114]: >>> 05:03:11:BC:00:10:81:5A
[... 40 lines cut ...]
[15:03:40][V][modbus_controller:190]: Range : 138F Size: 1 (3) skip: 100
[15:03:40][V][modbus_controller:190]: Range : 1391 Size: 1 (3) skip: 100
[15:03:40][V][modbus_controller:190]: Range : 139A Size: 1 (3) skip: 100
[15:03:40][V][modbus_controller:190]: Range : 139E Size: 1 (3) skip: 100
[15:03:40][V][modbus_controller:190]: Range : 13A0 Size: 1 (3) skip: 100
[15:03:41][V][modbus_controller:044]: Sending next modbus command to device 5 register 0x138A count 1
[15:03:41][V][modbus:210]: Modbus write: 05.03.13.8A.00.01.A0.E0 (8)
[15:03:41][V][modbus_controller:556]: Command sent 3 0x138A 1
[15:03:41][D][uart_debug:114]: >>> 05:03:13:8A:00:01:A0:E0
[15:03:41][V][sensor:043]: 'Load Power Factor': Received new state nan
[15:03:41][D][sensor:094]: 'Load Power Factor': Sending state nan  with 2 decimals of accuracy
[15:03:42][V][modbus_controller:044]: Sending next modbus command to device 5 register 0x138A count 1
[15:03:42][V][modbus:210]: Modbus write: 05.03.13.8A.00.01.A0.E0 (8)
[15:03:42][V][modbus_controller:556]: Command sent 3 0x138A 1
[15:03:42][D][uart_debug:114]: >>> 05:03:13:8A:00:01:A0:E0
[15:03:43][D][modbus_controller:040]: Modbus command to device=5 register=0x138A countdown=0 no response received - removed from send queue
[15:03:43][V][modbus_controller:044]: Sending next modbus command to device 5 register 0x138F count 1
[15:03:43][V][modbus:210]: Modbus write: 05.03.13.8F.00.01.B0.E1 (8)
[15:03:43][V][modbus_controller:556]: Command sent 3 0x138F 1
[15:03:43][D][uart_debug:114]: >>> 05:03:13:8F:00:01:B0:E1
[15:03:44][V][modbus_controller:044]: Sending next modbus command to device 5 register 0x138F count 1
[15:03:44][V][modbus:210]: Modbus write: 05.03.13.8F.00.01.B0.E1 (8)
[15:03:44][V][modbus_controller:556]: Command sent 3 0x138F 1
[15:03:44][D][uart_debug:114]: >>> 05:03:13:8F:00:01:B0:E1
```

sevrugin, 2024-06-29:
On HA I can see only Controller Temperature, RSSI and Uptime

sevrugin, 2024-06-29:
I'll answer by myself - RX/TX was swapped between RS232 and M5

korishan, 2024-08-15:
> I'll answer by myself - RX/TX was swapped between RS232 and M5

I'm getting this too. When you say swapped, do you mean on the left side of this chip??
[image]

Is this pinout diagram correct, or you had them soldered backwards?

sevrugin, 2024-08-15:
Try to reverse Rx and tx pins

korishan, 2024-08-15:
I've done that several times on the TTL/ESP32 side. nothing changed. I was wondering if the TTL/Inverter side that might need to be flipped

dimonezz, 2024-09-10:
Also very interesting, I use POW-HVM3.2K-24V and also get "unknown" for almost all values. I changed the wires, changed the TTL-RS converter.

diybateriaslifepo4, 2024-11-26:
Has anyone managed to get anything? I'm in the same situation and I can't get it to communicate no matter what I do.

ericsboro, 2024-11-26:
> Has anyone managed to get anything? I'm in the same situation and I can't get it to communicate no matter what I do.

I had same issue, but it was caused by uart-ttl module. if you soldered rx-tx wires from bottom side -- resolder on top, or vise versa. I noticed that on some modules VCC and GND pads are connected through the pcb, but rx and tx not. 
so try it :)

diybateriaslifepo4, 2024-11-26:
> I had same issue, but it was caused by uart-ttl module. if you soldered rx-tx wires from bottom side -- resolder on top, or vise versa. I noticed that on some modules VCC and GND pads are connected [...]

I am using the same converter and the same face as in the example photo, I am using the same face for input and output because I knew that in these converters the two faces are different channels but not that one was connected and the other was not.

diybateriaslifepo4, 2024-11-26:
> > > > Has anyone managed to get anything? I'm in the same situation and I can't get it to communicate no matter what I do.
> > > 
> > > 
> > > I had same issue, but it was caused by uart-ttl module. if you soldered rx-tx wires from bottom side -- resolder on top, or vise versa. I noticed that on some modules VCC and GND pads are connected through the pcb, but rx and tx not. so try it :)
> > 
> > 
> > I am using the same converter and the same face as in the example photo, I am using the same face for input and output because I knew that in these converters the two faces are different channels but not that one was connected and the other was not.
> 
> I have exactly visually the same as on photo, but it was not working with bottom soldering. after resoldering rx/tx to top pads it works like a charm :)

I changed to a completely different adapter and it still didn't work, but I found another project that did work:

https://github.com/syssi/esphome-smg-ii

So I understand that it communicates in a different way than this project uses

Thank you very much for trying to help.