odya/esphome-powmr-hybrid-inverter #55: Not Getting any data returned from inverter

korishan, 2024-08-15:
I have an ESP-WROOM-32D, the RS232->TTL shown in the documents and a PowMr 5kW hybrid inverter. I can communicate with the USB connection with the Windows app, so I know at least the inverter is outputting data.

I wired up the components as shown in the instructions and hooked it up to the unit. During boot up, I will get this message:
`[W][modbus_controller:027]: Modbus device=5 set offline`

I'm guessing this may be why it won't retrieve any data. I don't know how to put it into 'online' status. 

Here's the following debug log entry:
`[22:32:09][I][app:100]: ESPHome version 2024.7.3 compiled on Aug 14 2024, 22:31:31
[22:32:09][I][app:102]: Project odya.esphome-powmr-hybrid-inverter version 1.3.3
[22:32:09][C][wifi:599]: WiFi:
[22:32:09][C][wifi:427]:   Local MAC: 30:AE:A4:97:B6:70
[22:32:09][C][wifi:432]:   SSID: [redacted]
[22:32:09][C][wifi:435]:   IP Address: 192.168.2.124
[22:32:09][C][wifi:439]:   BSSID: [redacted]
[22:32:09][C][wifi:440]:   Hostname: 'powmr-inverter'
[22:32:09][C][wifi:442]:   Signal strength: -35 dB ▂▄▆█
[22:32:09][V][wifi:444]:   Priority: 0.0
[22:32:09][C][wifi:446]:   Channel: 10
[22:32:09][C][wifi:447]:   Subnet: 255.255.255.0
[22:32:09][C][wifi:448]:   Gateway: 192.168.2.1
[22:32:09][C][wifi:449]:   DNS1: 192.168.2.10
[22:32:09][C][wifi:450]:   DNS2: 192.168.2.1
[22:32:09][V][modbus_controller:044]: Sending next modbus command to device 5 register 0x1399 count 1
[22:32:09][V][modbus:210]: Modbus write: 05.03.13.99.00.01.51.25 (8)
[22:32:09][V][modbus_controller:557]: Command sent 3 0x1399 1
[22:32:09][C][logger:185]: Logger:
[22:32:09][C][logger:186]:   Level: VERBOSE
[22:32:09][C][logger:188]:   Log Baud Rate: 0
[22:32:09][C][logger:189]:   Hardware UART: UART0
[22:32:09][C][logger:193]:   Level for 'component': ERROR
[22:32:09][C][logger:193]:   Level for 'modbus_controller': VERBOSE
[22:32:09][C][logger:193]:   Level for 'modbus_controller.text_sensor': WARN
[22:32:09][C][logger:193]:   Level for 'modbus_controller.sensor': WARN
[22:32:09][C][logger:193]:   Level for 'modbus_controller.binary_sensor': WARN
[22:32:09][C][logger:193]:   Level for 'modbus_controller.select': WARN
[22:32:09][C][uart.idf:159]: UART Bus 0:
[22:32:09][C][uart.idf:160]:   TX Pin: GPIO17
[22:32:09][C][uart.idf:161]:   RX Pin: GPIO16
[22:32:09][C][uart.idf:163]:   RX Buffer Size: 256
[22:32:09][C][uart.idf:165]:   Baud Rate: 2400 baud
[22:32:09][C][uart.idf:166]:   Data Bits: 8
[22:32:09][C][uart.idf:167]:   Parity: NONE
[22:32:09][C][uart.idf:168]:   Stop bits: 1
[22:32:09][C][uart.idf:159]: UART Bus 1:
[22:32:09][C][uart.idf:160]:   TX Pin: GPIO1
[22:32:09][C][uart.idf:161]:   RX Pin: GPIO3
[22:32:09][C][uart.idf:163]:   RX Buffer Size: 256
[22:32:09][C][uart.idf:165]:   Baud Rate: 9600 baud
[22:32:09][C][uart.idf:166]:   Data Bits: 8
[22:32:09][C][uart.idf:167]:   Parity: NONE
[22:32:09][C][uart.idf:168]:   Stop bits: 1
[22:32:09][D][uart_debug:114]: >>> 05:03:13:99:00:01:51:25
[22:32:09][C][modbus:152]: Modbus:
[22:32:09][C][modbus:154]:   Send Wait Time: 250 ms
[22:32:09][C][modbus:155]:   CRC Disabled: NO
[22:32:09][C][modbus:152]: Modbus:
[22:32:09][C][modbus:154]:   Send Wait Time: 250 ms
[22:32:09][C][modbus:155]:   CRC Disabled: NO
[22:32:09][C][template.sensor:022]: Template Sensor 'PV Current'
[22:32:09][C][template.sensor:022]:   Device Class: 'current'
[22:32:09][C][template.sensor:022]:   State Class: 'measurement'
[22:32:09][C][template.sensor:022]:   Unit of Measurement: 'A'
[22:32:09][C][template.sensor:022]:   Accuracy Decimals: 1
[22:32:09][C][template.sensor:022]:   Icon: 'mdi:solar-power'
[22:32:09][C][template.sensor:023]:   Update Interval: 60.0s
[22:32:09][C][template.sensor:022]: Template Sensor 'Battery Current'
[22:32:09][C][template.sensor:022]:   Device Class: 'current'
[22:32:09][C][template.sensor:022]:   State Class: 'measurement'
[22:32:09][C][template.sensor:022]:   Unit of Measurement: 'A'
[... 246 lines cut ...]
[22:32:23][V][modbus_controller:557]: Command sent 3 0x11BC 16
[22:32:23][D][sensor:094]: 'Battery Current': Sending state nan A with 1 decimals of accuracy
[22:32:23][D][uart_debug:114]: >>> 05:03:11:BC:00:10:81:5A
[22:32:24][D][modbus_controller:040]: Modbus command to device=5 register=0x11BC countdown=0 no response received - removed from send queue
[22:32:24][V][modbus_controller:044]: Sending next modbus command to device 5 register 0x138A count 1
[22:32:24][V][modbus:210]: Modbus write: 05.03.13.8A.00.01.A0.E0 (8)
[22:32:24][V][modbus_controller:557]: Command sent 3 0x138A 1
[22:32:24][D][uart_debug:114]: >>> 05:03:13:8A:00:01:A0:E0
[22:32:25][V][modbus_controller:044]: Sending next modbus command to device 5 register 0x138A count 1
[22:32:25][V][modbus:210]: Modbus write: 05.03.13.8A.00.01.A0.E0 (8)
[22:32:25][V][modbus_controller:557]: Command sent 3 0x138A 1
[22:32:25][D][uart_debug:114]: >>> 05:03:13:8A:00:01:A0:E0
[22:32:26][V][modbus_controller:044]: Sending next modbus command to device 5 register 0x138A count 1
[22:32:26][V][modbus:210]: Modbus write: 05.03.13.8A.00.01.A0.E0 (8)
[22:32:26][V][modbus_controller:557]: Command sent 3 0x138A 1
[22:32:26][D][uart_debug:114]: >>> 05:03:13:8A:00:01:A0:E0
[22:32:26][V][sensor:043]: 'Load Current': Received new state nan
[22:32:26][D][sensor:094]: 'Load Current': Sending state nan A with 3 decimals of accuracy
[22:32:27][V][modbus:210]: Modbus write: 01.04.00.00.00.0A.70.0D (8)
[22:32:27][V][modbus_controller:044]: Sending next modbus command to device 5 register 0x138A count 1
[22:32:27][V][modbus:210]: Modbus write: 05.03.13.8A.00.01.A0.E0 (8)
[22:32:27][V][modbus_controller:557]: Command sent 3 0x138A 1
[22:32:27][D][uart_debug:114]: >>> 05:03:13:8A:00:01:A0:E0`

Not sure how much more I need to include.

korishan, 2024-08-15:
this seems off, I know the soc isn't that hot: Controller Temperature': Sending state **77.22222 °C** with 1 decimals of accuracy
So maybe there's more going on here than just not communicating?

korishan, 2024-08-15:
powmr-inverter.yaml:
substitutions:
  device_name: "powmr-inverter"
  node_name: "PowMr Inverter 1"
  node_id: powmr_inverter
  inverter_tx_pin: GPIO16
  inverter_rx_pin: GPIO17
  inverter_voltage_offset: "0"
  pzem_tx_pin: GPIO1
  pzem_rx_pin: GPIO3
  pzem_voltage_offset: "1"
  pzem_current_offset: "0.04"
  update_interval: 15s # Minimum 10s recommended to avoid duplicate command errors
  select_skip_updates: "2"
  pzem_update_interval: 5s # Needs to be 1/5 of update interval because of smoothing filter on pzem readings

esp32:
  board: esp32dev
  framework:
    type: esp-idf

packages:
  main: !include powmr-inverter/main.yaml
  common_defaults: !include includes/defaults.yaml
  common_sensors: !include powmr-inverter/common_sensors.yaml

logger:
  level: VERBOSE
  logs:
    component: ERROR # Fix for issue #4717 "Component xxxxxx took a long time for an operation"
    modbus_controller: VERBOSE
    modbus_controller.text_sensor: WARN
    modbus_controller.sensor: WARN
    modbus_controller.binary_sensor: WARN
    modbus_controller.select: WARN

======================
inverter.yaml:

uart:
  - id: uart_inverter
    baud_rate: 2400
    tx_pin: ${inverter_tx_pin}
    rx_pin: ${inverter_rx_pin}
    debug:
      direction: BOTH
      dummy_receiver: false

modbus:
  - id: modbus_inverter
    uart_id: uart_inverter
    send_wait_time: 250ms

modbus_controller:
  - id: smg_inverter
    address: 0x05
    modbus_id: modbus_inverter
    setup_priority: -10
    offline_skip_updates: 100
    command_throttle: 1s
    update_interval: ${update_interval}
......

dimonezz, 2024-09-10:
The problem is the same, I use the POW-HVM3.2H-24V inverter. The data I receive is Temperature Controller, RSSI, Uptime. All other values ​​are "unknown".

dimonezz, 2024-09-10:
> @dimonezz проверьте проводку, похоже, нет подключения UART, модель 3.2 должна работать

I looked at the wires, swapped them, changed the TTL-RS converter, I'll look again. Thank you

kevinari26, 2024-12-20:
is it works now?

kdmutriy, 2025-01-01:
very interesting problem. everything worked for me. assembled another device connected it to my inverter and it didn't work. returned the working one and now it doesn't work either