odya/esphome-powmr-hybrid-inverter #59: inverter+ BMS

fastvd, 2024-09-27:
Greetings. I have an esp32 board that works with jk-bms via modbus with the following firmware:
github://syssi/esphome-jk-bms@main

and also your great project i have successfully connected my POW-HVM10.2M inverter on separate esp32 board...

the idea is elementary: to control the inverter and bms from one esp32 board...

I seem to have combined the code of 2 firmwares, soldered everything correctly, but in the logs I get errors on the BMS part (the data from the inverter is fine)

here is my code:

**substitutions:
  name: bms-home
  device_name: "powmr-inverter"
  node_name: "PowMr Inverter"
  node_id: powmr_inverter
  inverter_tx_pin: GPIO1
  inverter_rx_pin: GPIO3
  inverter_voltage_offset: "0"
  
  bms_tx_pin: GPIO16
  bms_rx_pin: GPIO17

  pzem_tx_pin: GPIO9
  pzem_rx_pin: GPIO10
  pzem_voltage_offset: "1"
  pzem_current_offset: "0.04"
  update_interval: 30s # Minimum 10s recommended to avoid duplicate command errors
  select_skip_updates: "2"
  pzem_update_interval: 6s # Needs to be 1/5 of update interval because of smoothing filter on pzem readings

esp32:
  board: nodemcu-32s
  framework:
    type: esp-idf

packages:
  main: !include powmr-inverter/main.yaml
  # You can replace next 2 lines with your common configs for all devices
  common_system: !include includes/common_system.yaml
  common_sensors: !include includes/common_sensors.yaml

logger:
  level: WARN
  logs:
    component: ERROR # Fix for issue #4717 "Component xxxxxx took a long time for an operation"

time:
  - platform: homeassistant
    id: hass_time

#web_server:
#  port: 80
#  ota: false**

and uart part:

external_components:
  - source: github://syssi/esphome-jk-bms@main
    refresh: 0s
    
**uart:
  - id: uart_bms
    baud_rate: 115200
    rx_buffer_size: 384
    tx_pin: ${bms_tx_pin}
    rx_pin: ${bms_rx_pin}

jk_modbus:
  - id: modbus_bms
    uart_id: uart_bms
    rx_timeout: 50ms

jk_bms:
  - id: bms0
    jk_modbus_id: modbus_bms
    update_interval: 5s

binary_sensor:
  - platform: jk_bms
    balancing:
      name: "${name} balancing"
    balancing_switch:
      name: "${name} balancing switch"
    charging:
      name: "${name} charging"
    discharging:
      name: "${name} discharging"
    dedicated_charger_switch:
      name: "${name} dedicated charger switch"
    online_status:
      name: "${name} online status"

sensor:
  - platform: jk_bms
    min_cell_voltage:
      name: "${name} min cell voltage"
    max_cell_voltage:
      name: "${name} max cell voltage"
    min_voltage_cell:
      name: "${name} min voltage cell"
    max_voltage_cell:
      name: "${name} max voltage cell"
    delta_cell_voltage:
      name: "${name} delta cell voltage"
    average_cell_voltage:
      name: "${name} average cell voltage"
    cell_voltage_1:
      name: "${name} cell voltage 1"
    cell_voltage_2:
      name: "${name} cell voltage 2"
    cell_voltage_3:
      name: "${name} cell voltage 3"
    cell_voltage_4:
      name: "${name} cell voltage 4"
    cell_voltage_5:
      name: "${name} cell voltage 5"
    cell_voltage_6:
      name: "${name} cell voltage 6"
    cell_voltage_7:
      name: "${name} cell voltage 7"
    cell_voltage_8:
      name: "${name} cell voltage 8"
    cell_voltage_9:
      name: "${name} cell voltage 9"
    cell_voltage_10:
      name: "${name} cell voltage 10"
    cell_voltage_11:
      name: "${name} cell voltage 11"
    cell_voltage_12:
      name: "${name} cell voltage 12"
    cell_voltage_13:
      name: "${name} cell voltage 13"
    cell_voltage_14:
      name: "${name} cell voltage 14"
    cell_voltage_15:
      name: "${name} cell voltage 15"
    cell_voltage_16:
      name: "${name} cell voltage 16"

    power_tube_temperature:
      name: "${name} power tube temperature"
    temperature_sensor_1:
      name: "${name} temperature sensor 1"
    temperature_sensor_2:
      name: "${name} temperature sensor 2"
    total_voltage:
      name: "${name} total voltage"
    current:
      name: "${name} current"
    power:
      name: "${name} power"
    charging_power:
      name: "${name} charging power"
    discharging_power:
      name: "${name} discharging power"
    capacity_remaining:
      name: "${name} capacity remaining"
    capacity_remaining_derived:
      name: "${name} capacity remaining derived"
    temperature_sensors:
      name: "${name} temperature sensors"
    charging_cycles:
      name: "${name} charging cycles"
    total_charging_cycle_capacity:
      name: "${name} total charging cycle capacity"
    battery_strings:
      name: "${name} battery strings"
    errors_bitmask:
      name: "${name} errors bitmask"
    operation_mode_bitmask:
      name: "${name} operation mode bitmask"
    total_voltage_overvoltage_protection:
      name: "${name} total voltage overvoltage protection"
    total_voltage_undervoltage_protection:
      name: "${name} total voltage undervoltage protection"
    cell_voltage_overvoltage_protection:
      name: "${name} cell voltage overvoltage protection"
    cell_voltage_overvoltage_recovery:
      name: "${name} cell voltage overvoltage recovery"
    cell_voltage_overvoltage_delay:
      name: "${name} cell voltage overvoltage delay"
    cell_voltage_undervoltage_protection:
      name: "${name} cell voltage undervoltage protection"
    cell_voltage_undervoltage_recovery:
      name: "${name} cell voltage undervoltage recovery"
    cell_voltage_undervoltage_delay:
      name: "${name} cell voltage undervoltage delay"
    cell_pressure_difference_protection:
      name: "${name} cell pressure difference protection"
    discharging_overcurrent_protection:
      name: "${name} discharging overcurrent protection"
    discharging_overcurrent_delay:
      name: "${name} discharging overcurrent delay"
    charging_overcurrent_protection:
      name: "${name} charging overcurrent protection"
    charging_overcurrent_delay:
      name: "${name} charging overcurrent delay"
    balance_starting_voltage:
      name: "${name} balance starting voltage"
    balance_opening_pressure_difference:
      name: "${name} balance opening pressure difference"
    power_tube_temperature_protection:
      name: "${name} power tube temperature protection"
    power_tube_temperature_recovery:
      name: "${name} power tube temperature recovery"
    temperature_sensor_temperature_protection:
      name: "${name} temperature sensor temperature protection"
    temperature_sensor_temperature_recovery:
      name: "${name} temperature sensor temperature recovery"
    temperature_sensor_temperature_difference_protection:
      name: "${name} temperature sensor temperature difference protection"
    charging_high_temperature_protection:
      name: "${name} charging high temperature protection"
    discharging_high_temperature_protection:
      name: "${name} discharging high temperature protection"
    charging_low_temperature_protection:
      name: "${name} charging low temperature protection"
    charging_low_temperature_recovery:
      name: "${name} charging low temperature recovery"
    discharging_low_temperature_protection:
      name: "${name} discharging low temperature protection"
    discharging_low_temperature_recovery:
      name: "${name} discharging low temperature recovery"
    total_battery_capacity_setting:
      name: "${name} total battery capacity setting"
    current_calibration:
      name: "${name} current calibration"
    device_address:
      name: "${name} device address"
    sleep_wait_time:
      name: "${name} sleep wait time"
    alarm_low_volume:
      name: "${name} alarm low volume"
    manufacturing_date:
      name: "${name} manufacturing date"
    total_runtime:
      name: "${name} total runtime"
#    start_current_calibration:
#      name: "${name} start current calibration"
    actual_battery_capacity:
      name: "${name} actual battery capacity"
#    protocol_version:
#      name: "${name} protocol version"

switch:
  - platform: jk_bms
    charging:
      name: "${name} charging"
    discharging:
      name: "${name} discharging"

text_sensor:
  - platform: jk_bms
    errors:
      name: "${name} errors"
    operation_mode:
      name: "${name} operation mode"
    battery_type:
      name: "${name} battery type"
    password:
      name: "${name} password"
    device_type:
      name: "${name} device type"
    software_version:
      name: "${name} software version"
    manufacturer:
      name: "${name} manufacturer"
    total_runtime_formatted:
      name: "${name} total runtime formatted"**

zingreen, 2024-09-27:
This is a perfect idea. I tried to combine these projects, but there were also errors. I would really like to have such an implementation

odya, 2024-11-03:
- I'll try to do this) I think it will be better idea to move BMS related code to separate package and connect in main.yaml
- Main usability problem is hardware board that was designed to get 12v power from inverter. If you'll use it as BMS gateway it's better to connect board power input directly to battery. Fortunately, XL4005 suppurts up to 32v input so we can connect input to BMS out on 12v/24v batteries. But be careful, **don't forget to disconnect-tweak-reconnect XL4005 board** and **cut connection to invertor 12v power on RG45**

Latyn4ik, 2024-11-03:
It’s not necessary to connect to the BMS with wires. You can use BLE instead, right?

Latyn4ik, 2024-11-04:
great idea to mix two components in one esp

odya, 2024-11-04:
> It’s not necessary to connect to the BMS with wires. You can use BLE instead, right?

yep. You could use BLE to communicate with BMS. But you need to power controller somehow. If you keeping your inverter always on that will not be a problem. But if inverter becomes off controller shuts down too so its better to take power from battery instead of inverter, in this case

cahbegit, 2025-02-03:
Here is my code, the inverter connects via UART, everything is fine, but there is no Bluetooth connection with the BMS

```
external_components:
  - source: github://syssi/esphome-jk-bms@main
    refresh: always

substitutions:
  name: "jkbms"
  device_name: "powmrbms"
  node_name: "powmrbms"
  node_id: powmrbms
  inverter_tx_pin: GPIO20
  inverter_rx_pin: GPIO21
  inverter_voltage_offset: "0"
  update_interval: 30s # 
  select_skip_updates: "2"
  jk_bms_name: "jkbms"
  jk_bms_description: "JKBMSble"
  jk_bms_mac_address: "C8:47:80:1F:73:**"  
  jk_bms_protocol_version: "JK02_32S"

esp32:
  board: nodemcu-32s
  framework:
    type: esp-idf
    sdkconfig_options:
      CONFIG_ESP32_WIFI_SW_COEXIST_ENABLE: y  

packages:
  main: !include powmrbms/main.yaml
  # You can replace next 2 lines with your common configs for all devices
  common_system: !include powmrbms/common_system.yaml
  common_sensors: !include powmrbms/common_sensors.yaml

logger:
  level: WARN
  logs:
    component: ERROR # Fix for issue #4717 "Component xxxxxx took a long time for an operation"

time:
  - platform: homeassistant
    id: hass_time

web_server:
  port: 80
#  ota: false

esp32_ble_tracker:
  on_ble_advertise:
    then:
      - lambda: |-
          if (x.get_name().rfind("JK-", 0) == 0) {
            ESP_LOGI("ble_adv", "New JK-BMS found");
            ESP_LOGI("ble_adv", "  Name: %s", x.get_name().c_str());
            ESP_LOGI("ble_adv", "  MAC address: %s", x.address_str().c_str());
            ESP_LOGD("ble_adv", "  Advertised service UUIDs:");
            for (auto uuid : x.get_service_uuids()) {
              ESP_LOGD("ble_adv", "    - %s", uuid.to_string().c_str());
            }
          }

ble_client:
  - mac_address: ${jk_bms_mac_address}
    id: client0

jk_bms_ble:
  - ble_client_id: client0
    protocol_version: ${jk_bms_protocol_version}
    throttle: 5s
    id: bms0

binary_sensor:
  - platform: jk_bms_ble
    balancing:
      name: "${name} balancing"
   ***
```
I enabled the option "CONFIG_ESP32_WIFI_SW_COEXIST_ENABLE: y" because without it, the board doesn't catch any Wi-Fi signal at all.
Currently, for the BMS, I have a separate ESP32-C3-SuperMini board with an external antenna, which catches both Wi-Fi and Bluetooth without this option, but I can't upload any firmware via UART to it—the board immediately goes into a constant reboot loop, and I haven't been able to fix this.
Can you tell me—have you managed to run the Inverter and BMS on a single ESP32 board?