odya/esphome-powmr-hybrid-inverter #80: ARNEN MAX 10200 Inverter read PV1 and PV2 with ESP32

ribalki, 2025-05-23:
Hello,

I just read this post and I'm having the same problems. I also have an Arnen 10200 Max Hybrid inverter, and I can only read one PV, not the second.
I'm attaching the code in my YAML file. If you can help me, I've been looking for over 3 months without success. A copy of the YAML file would be ideal for me, as I'm not an ESP expert (just a beginner).

thanks in advance for your help.

> Unfortunately the registers don't seem to contain anything on my Anern AN-SCI-MAX-10200 that I bought in September 2023 either.
> 
> 
> 
> [image]
> 
> 
> 
> @bradmatt275 Do you see both strings on your inverter's display? 

 _Originally posted by @davidebeatrici in [#31](https://github.com/odya/esphome-powmr-hybrid-inverter/issues/31#issuecomment-2179124240)_

Mon fichier Yaml

# Warning:
#
# If you configure a lot of the possible sensors etc. it could be that you run
# out of memory (on esp8266). If you configure nearly all sensors etc. you run
# in a stack-size issue. In this case you have to increase stack size!
#
#  https://github.com/esphome/issues/issues/855

substitutions:
  name: pipsolar

esphome:
  name: ${name}
  min_version: 2024.6.0
  project:
    name: "syssi.esphome-pipsolar@pip8084"
    version: 1.0.0

esp32:
  # board: wemos_d1_mini32
  board: esp32dev
  framework:
    type: esp-idf

external_components:
  - source: github://syssi/esphome-pipsolar@pip8048
    refresh: 0s

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  ap:
    ssid: "Pipsolar Fallback Hotspot"
    password: "12345098"
    
ota:
  platform: esphome
  password: "7a32b08a138b4624c533156bf6a53650"

web_server:
  port: 80

logger:
  baud_rate: 0

mqtt:
  broker: 192.168.50.91              # Serveur MQTT
  username: usermqtt                 # User Serveur MQTT
  password: video                    # Password Serveur MQTT
[... 513 lines cut ...]
    on_value:
      then:
        - output.pipsolar.set_level:
            id: battery_float_voltage
            value: !lambda |- 
              float float_float = std::stof(id(set_float).state); // change to float number value
              return float_float;
    options: ["24.0","24.1","24.2","24.3","24.4","24.5","24.6","24.7","24.8","24.9","25.0","25.1","25.2","25.3","25.4","25.5","25.6","25.7","25.8","25.9","26.0","26.1","26.2","26.3","26.4","26.5","26.6","26.7","26.8","26.9","27.0","27.1","27.2","27.3","27.4","27.5","27.6","27.7","27.8","27.9","28.0","28.1","28.2","28.3","28.4","28.5","28.6","28.7","28.8","28.9","29.0","29.1","29.2","29.3","29.4","29.5","29.6","29.7","29.8","29.9","30.0","30.1","30.2","30.3","30.4","30.5","30.6","30.7","30.8","30.9","31.0","31.1","31.2","31.3","31.4","31.5","31.6","31.7","31.8","31.9","32.0","32.1","32.2","32.3","32.4","32.5","32.6","32.7","32.8","32.9","33.0","33.1","34.0","34.1","34.2","34.3","34.4","34.5","34.6","34.7","34.8","34.9","35.0","53.0","53.1","53.2","53.3","53.4","53.5","53.6","53.7","53.8","53.9","54.0","54.1","54.2","54.3","54.4","54.5","54.6","54.7","54.8","54.9","55.0","55.1","55.2","55.3","55.4","55.5","55.6","55.7","55.8","55.9","56.0","56.1","56.2","56.3","56.3","56.4","56.4","56.5","56.6","56.7","56.8","56.9","57.0","57.1","57.2","57.3","57.4","57.5","57.6","57.7","57.8","57.9","58.0","58.1","58.2","58.3","58.4","58.5","58.6","58.7","58.8","58.9"]

 
  - platform: template
    name: Battery - Voltage Back
    id: set_back
    optimistic: True
    initial_option: "45.0"
    restore_value: true
    entity_category: config
    on_value:
      then:
        - output.pipsolar.set_level:
            id: battery_back_voltage
            value: !lambda |- 
              float float_back = std::stof(id(set_back).state); // change to float number value
              return float_back;
    options: ["0.0","24.0","24.5","25.0","25.5","26.0","26.5","27.0","27.5","28.0","28.5","29.0","29.0","29.5","30.0","30.5","31.0","31.5","32.0","32.5","33.0","33.5","34.0","34.5","35.0","35.5","36.0","36.5","37.0","37.5","38.0","38.5","39.0","39.5","40.0","40.5","41.0","41.5","42.0","42.5","43.0","43.5","44.0","44.5","45.0","45.5","46.0","46.5","47.0","47.5","48.0","48.5","49.0","49.5","50.0","50.5","51.0","51.5","52.0","52.5","53.0","53.5","54.0","54.5","55.0"]

ribalki, 2025-06-03:
@bradmatt275 Do you see both strings on your inverter's display?
Yes, I have the both strings on the display

Central123454, 2025-12-31:
address: 4563
 address: 4564
These are the addresses of the current and voltage sensors, I added them and I got PV2 on POWMR